gaoxiang12/faster-lio #3: Test on Nvidia TX2

XINRUII, 2022-02-20:
Thanks for your newly released excellent work "Faster-Lio"! But I have some questions about the time consumption  on Nvidia TX2.   We have conducted several experiments with Zvision ML30SB1 Lidar,  and MTi-3DK imu.
Here are the results: 

I0220 23:57:42.142266 19196 utils.h:52] >>> ===== Printing run time =====
I0220 23:57:42.142282 19196 utils.h:54] > [     IVox Add Points ] average time usage: 5.36792 ms , called times: 7
I0220 23:57:42.142361 19196 utils.h:54] > [     Incremental Mapping ] average time usage: 16.9356 ms , called times: 7
I0220 23:57:42.142395 19196 utils.h:54] > [     ObsModel (IEKF Build Jacobian) ] average time usage: 9.74618 ms , called times: 28
I0220 23:57:42.142417 19196 utils.h:54] > [     ObsModel (Lidar Match) ] average time usage: 24.7205 ms , called times: 28
I0220 23:57:42.142438 19196 utils.h:54] > [ Downsample PointCloud ] average time usage: 0.524399 ms , called times: 7
I0220 23:57:42.142462 19196 utils.h:54] > [ IEKF Solve and Update ] average time usage: 174.88 ms , called times: 7
I0220 23:57:42.142488 19196 utils.h:54] > [ Preprocess (Standard) ] average time usage: 30.2425 ms , called times: 12
I0220 23:57:42.142508 19196 utils.h:54] > [ Undistort Pcl ] average time usage: 235.434 ms , called times: 8
I0220 23:57:42.142529 19196 utils.h:59] >>> ===== Printing run time end =====

From the experimental results, we can see that faster-lio cannot achieve 10hz on tx2. But Fast-lio2 usually consumes 50ms in our test with same down sampling rate. We guess it is because the system architecture is different and the relevant Intel acceleration libraries are not well adapted on the arm architecture.  How should we improve the efficiency of running on tx2?    Thanks a lot！

gaoxiang12, 2022-02-21:
I think maybe it's an intel tbb issue. I'll try to find a tx2 or Xavier to run a test.

gaoxiang12, 2022-02-21:
Could you double-check that the program is compiled in release mode and with -O2 flag? It seems the simple undistort pcl function has an abnormally long time usage.

XINRUII, 2022-02-21:
> 
Thanks for your reply!
We have rebuilt the program with -O2, but still get the same problem.

gaoxiang12, 2022-02-21:
Do you use plain cmake or catkin_make?

XINRUII, 2022-02-21:
We use catkin_make and here are the settings :

`cmake_minimum_required(VERSION 3.5)
project(faster_lio)

set(DEFAULT_BUILD_TYPE "Release")

set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_EXTENSIONS OFF)
set(CMAKE_CXX_STANDARD 17)

set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -pthread -fexceptions -g -ggdb")
set(CMAKE_CXX_FLAGS_RELEASE "-O3 ${CMAKE_CXX_FLAGS}")
set(CMAKE_CXX_FLAGS_DEBUG "${CMAKE_CXX_FLAGS}")

add_definitions(-DROOT_DIR=\"${CMAKE_CURRENT_SOURCE_DIR}/\")
#add_subdirectory(thirdparty/livox_ros_driver)

include(cmake/packages.cmake)

#definitions
add_definitions(-DROOT_DIR=\"${CMAKE_CURRENT_SOURCE_DIR}/\")

# options
option(WITH_IVOX_NODE_TYPE_PHC "Use PHC instead of default ivox node" OFF)

if (WITH_IVOX_NODE_TYPE_PHC)
    message("USING_IVOX_NODE_TYPE_PHC")
    add_definitions(-DIVOX_NODE_TYPE_PHC)
else ()
    message("USING_IVOX_NODE_TYPE_DEFAULT")
endif()

add_subdirectory(src)
add_subdirectory(app)
`

XINRUII, 2022-02-21:
We use catkin_make:

`cmake_minimum_required(VERSION 3.5)
project(faster_lio)

set(DEFAULT_BUILD_TYPE "Release")

set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_EXTENSIONS OFF)
set(CMAKE_CXX_STANDARD 17)

set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -pthread -fexceptions -g -ggdb")
set(CMAKE_CXX_FLAGS_RELEASE "-O3 ${CMAKE_CXX_FLAGS}")
set(CMAKE_CXX_FLAGS_DEBUG "${CMAKE_CXX_FLAGS}")

add_definitions(-DROOT_DIR=\"${CMAKE_CURRENT_SOURCE_DIR}/\")`

gaoxiang12, 2022-02-21:
Could you please recompile with explicit build type flag like ```catkin_make -DCMAKE_BUILD_TYPE=Release```  ? Please delete the current build and devel directories in your catkin workspace.

XINRUII, 2022-02-21:
I have tried this method, but I still get the problem...

gaoxiang12, 2022-02-21:
I'm not very sure what happens with your environment. Here's a test from my colleague using 16 lines spinning lidar and Nvidia Xavier:
```bash
 >>> ===== Printing run time =====
 [     IVox Add Points ] average time usage: 0.443667 ms , called times: 503
 [     Incremental Mapping ] average time usage: 0.808046 ms , called times: 503
 [     ObsModel (Lidar Match) ] average time usage: 3.73778 ms , called times: 34135
 [ Downsample PointCloud ] average time usage: 0.172217 ms , called times: 8774
 [ IEKF Solve and Update ] average time usage: 17.2345 ms , called times: 8773
 [ Preprocess (Standard) ] average time usage: 0.197055 ms , called times: 8778
 >>> ===== Printing run time end =====
```
This looks reasonable to me. He uses a tbb library compiled from the source, which may be different from the library provided in this repository.

XINRUII, 2022-02-21:
My tbb library is also compiled from the source, and I will double-check my environmental setting on TX2.

XINRUII, 2022-02-21:
My tbb library is also compiled from the source, and I will double-check my environmental setting on TX2. 
Anyway,  thank you for your reply!     ：）

closed 2022-02-21 by XINRUII