gaoxiang12/faster-lio #7: trying to minimize or remove state randomness of the same bag.

Doodle1106, 2022-03-07:
Hi,

I can witness large state randomness(-0.3 ~ 0.3 degree) when running the same bag multiple times, especially for the extrinsic estimation of Tlb. What parameters can I tune to minimize this randomness?

Thanks for trying to help.
Thanks.

gaoxiang12, 2022-03-08:
Hi @Doodle1106 ,
Extrinsics optimization is usually turned off in Faster-LIO since it is somewhat unstable and depends on the shape of the trajectory. You can also try to turn it off to check if that could bring any differences.

Doodle1106, 2022-03-09:
@gaoxiang12 

Thanks for the quick reply! 

I am running on my own data and I tried to increase max_iteration above 3, but the overall performance drastically deteriorates, odometry looks twisted, ba and bg looks weird and Tlb remains at 1e-35.   Are these results expected?

gaoxiang12, 2022-03-11:
Is it behaving correctly if max_iteration is 3? If so I just fixed a logic bug yesterday, please update the code and try again.