gaoxiang12/faster-lio #9: ivox地图点到达一定的数量后程序崩溃

AFEICHINA, 2022-03-14:
高博：
    您好！
    我在测试长时间rosbag包时发现当ivox地图到达约33w个点后，程序就莫名的崩溃。我修改ivox类型为PHC，也有类似的情况产生(ivox地图点为37w个点后崩溃)。从我测试的情况来看，这个问题是必现的。
    我尝试修改ivox3d.h文件中Options结构体的capacity_值，把它减小为30w，地图的点上限被限制了，但是崩溃的情况还是发生。上述情况崩溃前的建图结果都是正常的。我在不同的硬件平台上都测试了一下，x86和arm都出现了这个情况。您那边可以看下这个问题吗？
[image]

gaoxiang12, 2022-03-14:
hello,

你能提供数据或者更多的日志信息吗？比如打开glog，或者以离线方式来运行数据包，看看崩溃发生在什么地方。
iVox本身是不太会崩溃的，我们自己在几百公里的数据集上测过，也从来没碰到ivox崩溃的情况。我猜测是什么地方点云数据出了点什么问题（为空，或者出了非法值一类的东西），你可以加一些log来看看。

AFEICHINA, 2022-03-14:
> hello,
> 
> 你能提供数据或者更多的日志信息吗？比如打开glog，或者以离线方式来运行数据包，看看崩溃发生在什么地方。 iVox本身是不太会崩溃的，我们自己在几百公里的数据集上测过，也从来没碰到ivox崩溃的情况。我猜测是什么地方点云数据出了点什么问题（为空，或者出了非法值一类的东西），你可以加一些log来看看。

感谢高博这么快的回复，我先检查下自采的数据问题，因为bag文件比较大，我按您说的方式先看看glog，后面我再反馈下进展。

AFEICHINA, 2022-03-14:
高博好，判断是自采数据包的问题。在只播包的情况下，使用rviz看原始点云数据，rviz同样也崩溃了。
[image]
查了下相关的问题确定是自采数据的错误，[https://blog.csdn.net/Electrical_IT/article/details/108243193](url)
再次感谢高博的热心回复。

closed 2022-03-14 by AFEICHINA