gaoxiang12/faster-lio #10: std::pair<_FIter, _FIter> MTK::cos_sinc_sqrt(const scalar&) [with scalar = double]: Assertion `x2 >= 0 && "argument must be non-negative"' failed

narutojxl, 2022-03-18:
Hello Dr. Gao and all authors,
Thanks for sharing this excellent work firstly. Downloaded [nclt bag: 20130110.bag]( https://drive.google.com/drive/folders/1VBK5idI1oyW0GC_I_Hxh63aqam3nocNK), `roslaunch faster_lio mapping_velodyne.launch`, crashed with the following infos. All the code are cloned with `git clone`.  How to run demo with nclt bag? Thanks for your help and time  a lot !
>
	faster-lio/include/IKFoM_toolkit/esekfom/../mtk/types/../src/mtkmath.hpp:157: std::pair<_FIter, _FIter> MTK::cos_sinc_sqrt(const scalar&) [with scalar = double]: Assertion `x2 >= 0 && "argument must be non-negative"' failed.
	[laserMapping-2] process has died [pid 835148, exit code -6, cmd /home/jxl/jxl_code/dlo_ws/devel/lib/faster_lio/run_mapping_online __name:=laserMapping __log:=/home/jxl/.ros/log/3ebb6572-a694-11ec-a628-ff1c388ae2a2/laserMapping-2.log].
	log file: /home/jxl/.ros/log/3ebb6572-a694-11ec-a628-ff1c388ae2a2/laserMapping-2*.log

gaoxiang12, 2022-03-23:
Hi @narutojxl , 
I just fixed this issue in the commit [6e95766](https://github.com/gaoxiang12/faster-lio/commit/6e957664f1a475305eecdb633df27749dd6902e9). This is caused by an incorrect std::move call in LaserMapping::ObsModel. Please test the program again.

narutojxl, 2022-03-23:
Thanks Dr. Gao :), close now.

closed 2022-03-23 by narutojxl