gaoxiang12/faster-lio #11: faster-lio迭代次数与点云残差收敛性

NEU-LC, 2022-03-27:
高博您好：
感谢faster-lio的Contribution。

有两个疑问想咨询一下：
1. 迭代次数设置为3的时候，我打印IEKF的循环次数发现很多帧在3次循环内很难收敛，迭代次数设置为10的时候发现有一些lidar帧4-5次可以收敛，但是即使是收敛了发现精度和迭代次数设置为3次也没有明显提高，这是因为什么。
2. 在计算点到平面残差的时候，和优化不一样的是，优化每迭代一次都计算一次残差，为什么fast-lio系列要每次收敛才要重新计算残差，为什么这样选取线性化点。

谢谢高博。

gaoxiang12, 2022-03-28:
1. IEKF在目前的设置下，avia数据集的平均迭代次数4-5次，nclt要7-8次才满足现在的收敛条件。我们自己做过一些实验也显示了这个问题。然而，如果仅是把最大迭代次数放开，那么IEKF可能得到更小的残差，但这和最终的轨迹精度不一定有直接关系。这仅能说明雷达点云在配准之后误差更小了，但不一定就更符合真值轨迹。真值轨迹通常是其他种类的传感器读到的，整套系统的标定误差是否足够小到可以敏感到这种程度的偏差，仍然是存疑的。
2. 这是fastLIO2里的一个trick，并不是不计算残差，只是不重新计算最近邻，残差还是会重新计算的。这种做法使得IEKF在4次迭代中实际只计算了2次最近邻，可以节省一些计算时间。

NEU-LC, 2022-03-28:
> 

感谢高博回复。关于点云前端匹配这里总想找一些办法来定性或者定量的评估一下准确性，感觉很多Lidar的算法在不同场景都是因为前端配准引起的，关于LOAM系列提取特征的方法和Fast-Lio系列直接配准的方法，是否有一些方法可以来衡量一下前端的准确性。之前我们就是把点到面的距离画出来看一下，但是看这种距离残差就像您说的这样好像不能说明前端配准的问题。

gaoxiang12, 2022-03-28:
在一些专门做registration文章里有点云的数据集，里边有仿真的点云数据和真实数据，那个匹配精度和lidar残差关系比较明显。像slam数据集的真值基本都是用RTK或动捕来做的，在很精细的尺度下，不一定就和点云的匹配误差有那么直接的关系。
或许你可以试试找找registration的数据集，我们自己也在做加快迭代的算法。

NEU-LC, 2022-03-28:
> 在一些专门做registration文章里有点云的数据集，里边有仿真的点云数据和真实数据，那个匹配精度和lidar残差关系比较明显。像slam数据集的真值基本都是用RTK或动捕来做的，在很精细的尺度下，不一定就和点云的匹配误差有那么直接的关系。 或许你可以试试找找registration的数据集，我们自己也在做加快迭代的算法。

好的，谢谢高博。

closed 2022-03-28 by NEU-LC

silent-zhou, 2024-01-30:
> > 在一些专门做registration文章里有点云的数据集，里边有仿真的点云数据和真实数据，那个匹配精度和lidar残差关系比较明显。像slam数据集的真值基本都是用RTK或动捕来做的，在很精细的尺度下，不一定就和点云的匹配误差有那么直接的关系。 或许你可以试试找找registration的数据集，我们自己也在做加快迭代的算法。
> 
> 好的，谢谢高博。

您好，我现在自己数据上也发现lio里程计不太好，想请问您有尝试过相关的registration文章里有点云的数据集数据集