gaoxiang12/faster-lio #14: 动态车辆导致SLAM退化

OliverShaoPT, 2022-04-19:
高博您好，
在测试您的算法的时候，发现当一辆视野内的静止车辆突然起步时，因为地图中已有该车辆的许多点云，使得SLAM前端倾向于将该车辆的动态点云匹配到历史的静态地图点上，导致前端退化，表现为原地不动。
想请问您，有没有不用深度学习语义信息的方法可以解决该问题？（如何判断退化出现？如何修改IMU与LIDAR的权重？）

gaoxiang12, 2022-04-21:
可以考虑融合轮速。轮速能够比较准确地反映车辆在前进方向上的状态。
我们实际工程中用的faster-lio还包括了RTK与轮速的融合，效果要稳定不少。

OliverShaoPT, 2022-04-21:
> 可以考虑融合轮速。轮速能够比较准确地反映车辆在前进方向上的状态。 我们实际工程中用的faster-lio还包括了RTK与轮速的融合，效果要稳定不少。

我在实验过程中，有尝试使用factor graph 的方法，在GTSAM 后端加入RTK数据，融合faster-lio的里程计和RTK定位。但因为RTK数据还是没有输入到SLAM系统中，起不到纠正SLAM前端漂移的效果。按照高博您说的方法，是不是需要把RTK和轮速加入前端的观测方程中呢？（求学习案例，直接看代码没有头绪）

gaoxiang12, 2022-04-21:
是的，直接加在观测方程中就行，需要一定程度上改写这边的ESKF代码。
RTK有旋转和平移观测后，还能防止LIO在零空间里的发散情况，体现为：1）引入RTK后LIO不会翘起来或者陷下去；2）LIO的加计零偏与重力方向不再耦合（没有绝对位姿观测时它们是耦合的，LIO跑久了可能出现很大的加计零偏和很奇怪的重力方向）。

OliverShaoPT, 2022-04-21:
感谢高博解答！

closed 2022-04-21 by OliverShaoPT