gaoxiang12/faster-lio #16: 融合RTK相关

shuqiu0202, 2022-04-29:
高博您好，

shuqiu0202, 2022-04-29:
[image]

我按照IKFoM的定义了一个测量方程h和两个偏导函数dh_dx,dh_dv，并使用update_iterated_dyn_share_modified，kf.update_iterated(z, R)进行rtk位置的测量更新，预测还是基于fast_lio2，现在看建模效果不好，会漂

xiaotaw, 2022-05-07:
IKFoM中用的应该是error state KF，看你的measure_model是按KF的方式来的，这样会出问题吧。

shuqiu0202, 2022-05-09:
@xiaotaw 我的理解是状态向量里面位置变量不是误差值，所以观测方程中的转换矩阵H中位置对应子矩阵应该是一个单位矩阵，另外状态预测值里面位置P属于R3，并非so(3)，而rtk给出的测量值也是一个位置，故它们之间的就是一个线性关系

xiaotaw, 2022-05-09:
你的说法没问题，既然你用kf的方式去使用RTK的观测量的话，那你就不能用IKFoM中的update_iterated函数。 如果你用IKFoM中的update_iterated函数的话，对应的观测应该使用eskf的方式，而不是kf的方式。

shuqiu0202, 2022-05-10:
@xiaotaw  对，您说的有道理，我应该是混用了，观测方程我是这样定义的：
[image]
现在不知道这个IKFoM中要求"output equation h(x, v)"该怎么写了，它需要实现一个measurementModel函数，请问有什么建议吗？我是否应该将rtk的位置测量值和预测状态x的位置的差值作为返回值？

xiaotaw, 2022-05-11:
可以参考一下官方的文档 https://github.com/hku-mars/IKFoM 和它的论文

shuqiu0202, 2022-05-11:
@xiaotaw  好的，谢谢，我再去看下

insistyxj, 2022-05-25:
> @xiaotaw ，谢谢，我再去下好了
您好，能否提供联系方式，有点问题请教一下

shuqiu0202, 2022-05-25:
> @shuqiu0202 您好，能否提供联系方式，有点问题请教一下

可以发邮件给我哈 shuqiu0202@163.com

closed 2022-08-18 by shuqiu0202