gaoxiang12/faster-lio #17: Question about visualize NCLT dataset

Taeyoung96, 2022-05-06:
@gaoxiang12  
Hi, Thanks for your great work!  
I have some questions about visualizing NCLT dataset with faster-lio.  
Actually, my question is not really related to faster-lio algorithm but visualizing the NCLT dataset's ground pose.  

I want to visualize the result plot like faster-lio paper,  

[image]

However, when I download the `groundtruth.csv` on the [original NCLT homepage](http://robots.engin.umich.edu/nclt/index.html#documentation),  
The data format is quite different from `traj.txt`(Estimated trajectory).  

How could I align `groundtruth.csv` and `traj.txt`?   
Any resources or tips to help align the timestamp and align with the GT and LiDAR coordinate system?   

Thanks,

xiaotaw, 2022-05-06:
hello, we convert `groundtruth.csv` into 'tum' format (`traj.txt` is already in 'tum' format), then use (evo)[https://github.com/MichaelGrupp/evo] to plot.

Taeyoung96, 2022-05-07:
@xiaotaw   
Thanks for your reply.  
Is there any recommended codes or links that convert `groundtruth.csv` to 'tum' format?  

I'm newbie so I have difficulty because `groundtruth.csv` is first format that I haven't seen yet. :(

xiaotaw, 2022-05-07:
There is no off-the-shelf code kept in my computer. It's very easy to write several python or c++ lines of code for this. Just read the trajectory from csv file and re-write them into a file in tum format. For tum format, here is a intro https://github.com/MichaelGrupp/evo/wiki/Formats#tum---tum-rgb-d-dataset-trajectory-format

Taeyoung96, 2022-05-09:
@xiaotaw   
Thanks, I'll try it by referring your link and NCLT original paper. 👍

closed 2022-05-09 by Taeyoung96