gaoxiang12/faster-lio #18: Question about creating the NCLT sample bag file.

Taeyoung96, 2022-05-11:
@xiaotaw  
Hi, I have some questions about [NCLT sample dataset](https://github.com/gaoxiang12/faster-lio#prepare-the-datasets).  

On the original [NCLT homepage](http://robots.engin.umich.edu/nclt/index.html#top), there are no official bag files but, `tar.gz` files.  
So I use [nclt2ros](https://github.com/bierschi/nclt2ros) repository to make the bag file.  

For example, when I make my own bag file for `2013-01-10`seuquence, I was able to create a ros topic like below.  

<p align="center">[image]</p>  

I execute the sequence(2013-01-10) and the [config file](https://github.com/gaoxiang12/faster-lio/blob/main/config/velodyne.yaml) (just change the topic name).  

And I got a weird result. :sob:   

<p align="center">[image]</p>   

The output of the terminal is below.
<p align="center">[image]</p>   

---  

And I exectue the same sequence with same [config file](https://github.com/gaoxiang12/faster-lio/blob/main/config/velodyne.yaml) with your sample Dataset, it works fine!  

On the [sample dataset](https://github.com/gaoxiang12/faster-lio#prepare-the-datasets), ros topic is like below.  

<p align="center">[image]</p>   
 
Result is like below.  

<p align="center">[image]</p>   

The output of the terminal is below.  
<p align="center">[image]</p>   

I think the difference between two terminals are   
```
Failed to find match for field 'time'
Failed to find match for field 'ring'
```   

What causes this result? Any help would be greatly appreciated.  

Thanks,

xiaotaw, 2022-05-17:
The sample dataset comes from FAST-LIO2, https://github.com/hku-mars/FAST_LIO#4-rosbag-example. Maybe, I am not sure, it differs from the one generated via nclt2ros.

Taeyoung96, 2022-05-17:
@xiaotaw   
Thanks for your reply. I will refer to the link to solve the problem.

closed 2022-05-17 by Taeyoung96