gaoxiang12/faster-lio #24: 如何进一步提升mapping效果？

pangchenglin, 2022-05-26:
高博你好：
[image]

           我在自己的数据集上测试faster-lio效果，发现在机器人静止时，存在这种情况：当前帧点云并不稳定，在来回跳动，保存的历史点云就存在一个长长的拖影，这个问题影响了建图的精度，导致墙壁会比较厚。但是我在查看fastlio和faster-lio视频的时候，发现建图效果还是不错的，当然也可能是一些细节没有看到，我想问下这个情况是什么因素引起的？有没有什么方法能够减小这个影响。

gaoxiang12, 2022-05-27:
LIO系统和纯点云系统不一样，对IMU有一定要求。IMU的轴和激光对齐了吗？不对齐的话需要改外参。

pangchenglin, 2022-05-27:
> 

用的是16线的ouster激光和ouster自带的imu，参数用的faster-lio例程里面64线的默认参数，除了外参之外是不是还需要自己标定一下imu的噪声？

gaoxiang12, 2022-05-29:
我不太确定ouster的具体情况，我手边只用过velodyne和avia的数据集，有可能是imu的噪声太大的问题。

rlabs-oss, 2022-06-24:
Do you have any suggestions when using a wit-motion IMU which already includes EKF? Ive tried this with the mid-40 lidar at 200Hz sample rate and 286hz filter (so no smoothing) - the scan lines and blind parameters are not known (ive tried 1 and 4), nor the acc_cov etc - what should those be to tune the response to an IMU? How were the Avia and Velodyne parameters derived? Faster_Lio will produce a scan that starts to rapidly drift making a long point cloud snake :/