gaoxiang12/faster-lio #25: Questions on use & mapping points in the global map with pose per file?

rlabs-oss, 2022-05-28:
Currently the trajectory is available but this is the whole global map trajectory. Is it possible to also dump the pose (stamp, tx,ty,tz, qx,qy,qz,qw) in another file for each individual .pcd file? (thus it would be possible to load the scans to rtab-map and experiment with loop closure) (or create a pose_01..n.csv file matching the .pcd last written)

Here's a scan made with the Avia mounted on a bicycle handlebar. Quite amazing considering the terrain varied from smooth tarmac to bumpy stones and gravel / dirt road. 

If a direction change is made (e.g turn around 180 degrees) within the current working memory are points more accurately registered? E.g is there any impact (besides memory) if the scan frame rate is set high so each .pcd file is e.g 100mb? The map below was made with "  interval: 200    # how many LiDAR frames saved in each pcd file;" but does this impact to poor registration or does it improve if returning to the same area? (I understand no loop closure is made) 

[image]

gaoxiang12, 2022-05-31:
Hi @rlabs-oss 

1. The trajectory is currently dumped in a txt file but not with a separate pcd of the corresponding scan file. The scans are merged together when the program quits. It would not be hard to write some lines to save the scan of each frame/keyframe since they are just the member variables of the EKF state struct. 

2. The rotation will be estimated by the IMU measurements so I think if your IMU is capable to observe the rotation, the point cloud can be correctly registered. But if the save option is opened, it may affect the registration since it takes time (maybe >100ms) to dump the point clouds into a disk file. New IMU measurements and point clouds would be skipped if the program is busy dumping those results.

3. If you return to a previously visited area (when the voxels are still active), the registration will take those points into consideration so there won't be any accuracy issues. It acts as a local map in many LIO systems. The size of the local map depends on the LRU size of iVox. And if you visit a too old area where the voxels are already cleaned, the map may show some double-wall artifacts because we don't have loop closing modules.

rlabs-oss, 2022-05-31:
Thank you, really interesting points to consider. 
for the first it would be good to use since it would allow import of the scans and corresponding poses to rtab-map & thus give the opportunity to experiment with loop closure offline in a stand-alone application.
For points 2 and 3, it was more z drift returning back to an area some 3-4km from the origin - this is expected given no loop closure - but I found the area at the bottom left of the image interesting where there was no double points. Ill disable intermediate saving and check how it improves the scan.
For the Avia lidar, was the device placed in repeating scan mode in a different application before using it with faster_lio? - currently my avia starts in non-repeating mode.

closed 2022-05-31 by rlabs-oss