gaoxiang12/faster-lio #28: Question about executing faster-lio with LIO-SAM dataset

Taeyoung96, 2022-06-08:
@gaoxiang12  
Hi, Thanks for your great work!  
I have 2 questions about faster-lio executing with Velodyne VLP 16.  
1. In [paper](https://ieeexplore.ieee.org/abstract/document/9718203), there is an experiment with LIO-SAM dataset.  

<p align="center">[image]</p>  

  On the LIO-SAM repository, some [sample datasets](https://github.com/TixiaoShan/LIO-SAM#sample-datasets) are provided.  
  Then, which sequence did you use to write a paper?  
    Do you make ground-truth trajectories with `gps/fix` topic in the sample dataset?  

2. I want to test Faster-lio algorithm with aggressive motion using Velodyne LiDAR.  
I find the simple dataset provided in [Rotation dataset in LIO-SAM](https://github.com/TixiaoShan/LIO-SAM#sample-datasets).  
But I got a wired result with provided `velodyne_liosam.yaml`.  
Actually, I fixed some parameters like below.  
```
common:
    dataset: "liosam"
    lid_topic:  "points_raw"
    imu_topic:  "imu_correct"
    time_sync_en: false         # ONLY turn on when external time synchronization is really not possible
    
preprocess:
    lidar_type: 2                # 1 for Livox serials LiDAR, 2 for Velodyne LiDAR, 3 for ouster LiDAR, 
    scan_line: 16
    blind: 4
    time_scale: 1e3             # Compatible with time units of different datasets

mapping:
    acc_cov: 3.9939570888238808e-03
    gyr_cov: 1.5636343949698187e-03
    b_acc_cov: 6.4356659353532566e-05
    b_gyr_cov: 3.5640318696367613e-05
    fov_degree: 180
    det_range: 100.0
    extrinsic_est_en: false      # true: enable the online estimation of IMU-LiDAR extrinsic
    extrinsic_T: [ 0, 0, 0]
    extrinsic_R: [ 1, 0, 0, 
                   0, 1, 0, 
                   0, 0, 1]

publish:
    path_publish_en:  false
    scan_publish_en:  true       # false: close all the point cloud output
    scan_effect_pub_en: true    # true: publish the pointscloud of effect point
    dense_publish_en: true       # false: low down the points number in a global-frame point clouds scan.
    scan_bodyframe_pub_en: true  # true: output the point cloud scans in IMU-body-frame

path_save_en: true                 # Save trajectories for accuracy calculations and comparisons

pcd_save:
    pcd_save_en: true
    interval: -1                 # how many LiDAR frames saved in each pcd file; 
                                 # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.

feature_extract_enable: false
point_filter_num: 3
max_iteration: 3
filter_size_surf: 0.5
filter_size_map: 0.5
cube_side_length: 1000

ivox_grid_resolution: 1.0        # default=0.2
ivox_nearby_type: 26             # 6, 18, 26
esti_plane_threshold: 0.1        # default=0.1
```

I shared my result video.  
- Rotation dataset with LIO-SAM - [[Link]()].  
- Rotation dataset with Faster-LIO - [[Link]()].  

Do I need to fix other parameters?  
Does it right that the faster-lio algorithm doesn't work well in this rotation dataset?  
(I guess correspondence issue or measurement model or direct method or what else...  :monocle_face:)  

I'll look forward to your reply.  
Thanks,

gaoxiang12, 2022-06-09:
@xiaotaw Did you or Bai Chunge do those LIO-SAM experiments? I'm not quite sure about the details here.

xiaotaw, 2022-06-10:
1.1 In the paper, `liosam_1` is short for `campus_small_dataset.bag`.
1.2 Yes, the groundtruth of LIO-SAM dataset comes from topic `/gps/fix`. As rotation is not provided, we filled them with zeros(i.e. the `qx qy qz qw` columns in `tum` file). And the groundtruth trajectory is converted into 100Hz by b-spline interpolation.

2 Due to the limited time, we did not tuned faster-lio for other  LIO-SAM's datasets. According to your video, it seems something goes wrong with the bias of the accelerometer, not sure yet.

Taeyoung96, 2022-06-11:
@xiaotaw   
Thanks for your reply. 😄    

Suppose the result is not good even if I change the parameters associated with the bias of the accelerometer.    
Could I understand that faster-lio is not an appropriate algorithm for the rotation dataset?

xiaotaw, 2022-06-13:
It could be more precise and rigorous to make such a conclusion, if you test on more datasets, and make sure that those datasets are properly fed into the algorithm without any other issues, such as timestamp, extrinsic calibration, etc.

Taeyoung96, 2022-06-13:
@xiaotaw   
Thanks for your advice. I'll run tests on the other datasets and look at other issues. :+1:

closed 2022-06-13 by Taeyoung96

xingyuuchen, 2022-12-06:
> 1.1 In the paper, `liosam_1` is short for `campus_small_dataset.bag`. 1.2 Yes, the groundtruth of LIO-SAM dataset comes from topic `/gps/fix`. As rotation is not provided, we just fill them with zeros(i.e. the `qx qy qz qw` columns in `tum` file). And the groundtruth trajectory is converted into 100Hz via b-spline interpolation.
> 
> 2 Due to the limited time, we did not tuned faster-lio for other LIO-SAM's datasets. According to your video, it seems something goes wrong with the bias of the accelerometer, not sure yet.

Hello,
A tiny problem in the paper:
The distance of `campus_small_dataset.bag` is only ~0.56km, rather than 1.43 km which is in the paper.
1.43km is for `campus_large_dataset.bag`.

Best,
Xingyu