gaoxiang12/faster-lio #35: 关于左乘扰动和右乘扰动

Torchmm, 2022-08-11:
高翔老师：
        您好，我在您的代码中ObsModel看到对旋转的求导用的好像是右乘扰动，在您的书上的推导看到是左乘扰动，有点不一样，我也看到说左乘扰动和右乘扰动都可以，但是得到的雅可比是不一样的。所以想请教一下：
1.为什么雅可比不一样的情况下，能得到差不多的优化结果？
2.是否左乘扰动和右乘扰动可以任意替换，或者在某些情况下用某种扰动会更好一些呢？

gaoxiang12, 2022-08-12:
hello,
1. 左乘要对应左乘的雅可比，右乘要对应右乘的雅可比，不是混着来的。如果对应上了，结果上没有本质差别。
2. 理论上都是一样的，使用上会有习惯差异。组合导航方面习惯用Twb表示，然后用右乘；CV领域则习惯Tcw表示，用左乘。

Torchmm, 2022-08-12:
谢谢高翔老师的回复。
1、是否可以理解成扰动的方向和状态叠加方向需要保持一致，如果是左乘扰动，那求解出来的δR需要左乘到原来的R上，右乘扰动则需要右乘到原来的R上。
2、我看到RP对R的的左乘扰动是-(RP)^，而RP对R的右乘扰动是-R(P)^，尝试修改了
[ObsModel函数里的A矩阵](https://github.com/gaoxiang12/faster-lio/blob/main/src/laser_mapping.cc#L641)
`common::V3F A(point_crossmat * C)`
改成
`common::V3F A(Rt.transpose()*point_crossmat * C)` 即把这里的右乘扰动改成左乘扰动。
这样和状态叠加方向应该不一致了，不过跑了一些数据集后发现，和没改的结果很类似，我没有去仔细的评估轨迹差异，不过看起来差不多。点云一致性看着也差不多。

gaoxiang12, 2022-08-15:
你的理解没有问题。

Torchmm, 2022-08-22:
好的，谢谢高翔老师！

closed 2022-08-22 by Torchmm