gaoxiang12/faster-lio #37: Arm architecture

engcang, 2022-08-31:
Hi @gaoxiang12 . Thanks for your great work and continuously managing issues.

I am trying to build faster-lio on Jetson Xavier NX and found your comment on other issue: https://github.com/gaoxiang12/faster-lio/issues/3#issuecomment-1046626652
Can you kindly let us know which version of `tbb`  you used? If you can, can you please give me the link?

thanks.

gaoxiang12, 2022-09-05:
Sorry I'm not quite sure about the detailed version in Xavier. @xiaotaw Could you help me check it?

xiaotaw, 2022-09-07:
Here is the version info that I found in a file named `CHANGES`
```
Intel TBB 2018
TBB_INTERFACE_VERSION == 10000
```

engcang, 2022-09-07:
@xiaotaw Thank you for sharing the information.

For the future users of this repository, I found that this modified `tbb` with version 2020.2 works well on both of ARM and AMD64.

https://github.com/syoyo/tbb-aarch64

Closing the issue.

closed 2022-09-07 by engcang