gaoxiang12/faster-lio #38: 可以，如果RTK和激光同步，那么就在观测方程中添加几行对R和T的观测即可，推导一下相应的雅可比阵。RTK单独观测也可以。

chengwei0427, 2022-09-05:
可以，如果RTK和激光同步，那么就在观测方程中添加几行对R和T的观测即可，推导一下相应的雅可比阵。RTK单独观测也可以。
不过原版FastLIO2基于MTK来表达状态变量，改起来可能麻烦一些。

_Originally posted by @gaoxiang12 in https://github.com/gaoxiang12/faster-lio/issues/13#issuecomment-1091105752_

chengwei0427, 2022-09-05:
高博@gaoxiang12 ，你好，
最近在考虑做`lio+间歇性gnss`的融合，有些问题想请教一下。
目前的做法是：

1. 采集一段时间的数据，离线对齐gnss和LIO轨迹，得到LIO到gnss坐标系下得变换矩阵；
2. 以步骤1得到的变换矩阵，作为LIO的初始化位姿（保证LIO和gnss坐标的粗对齐），之后进行融合gnss；
3. 存在lidar时，进行lidar观测更新，存在gnss时，进行gnss观测更新；
问题：
1. 由于是间歇性gnss，若一段时间后，再次观测到gnss之后，对lidar激光数据进行观测时，局部地图该怎么处理呢？测试发现，由于局部地图还是之前构造的，会导致轨迹有折线，甚至直接偏移？
2. 若gps和lidar激光帧同时进行观测，这种lidar和gnss的不同步（频率不同），有什么好的建议呢？

chengwei0427, 2022-09-06:
email中已解答，感谢高博的耐心回复。

再次感谢！

closed 2022-09-06 by chengwei0427

luameows, 2023-04-03:
@chengwei0427 想請教下，通過對齊gnss與lio軌跡後拿到的RT的精度應該不足以讓兩個odom軌跡完全對齊把。那此時，在觀測時，rtk和lidar兩個觀測的pose和rotation就有偏差，這時您如何處理呢？

Fafa-DL, 2023-04-13:
hello，遇到了同样的问题，请问高博当时是如何给出解决方案的呢？

chengwei0427, 2023-04-14:
Hi, @luameows @Fafa-DL 

```
1. LIO主要的残差还是雷达点云那块配准过程，顺便得到了IMU高维状态。由于紧耦合里雷达点云的观测方程数量要明显多于运动方程，整体系统应该很快向点云收敛，这使得LIO系统比较依赖点云形状。
2. 而RTK由于低频问题，每隔一段时间才作用一次（只改变那个时刻的Pose，对点云形状影响较小），而状态又主要取决于点云，所以应该要减小点云的影响（或者，如果历史位姿发生改变，点云地图也要跟着变化）。

我觉得需要处理局部地图。可以尝试一下几个思路：
a. 局部地图只在有GNSS的那个帧构建，其他帧的scan不进入局部地图，只做配准；
b. 或者，类似于msckf的思路，允许调整局部地图里那些scan的位姿，但那样的话可能不适合用ivox来实现。

Dr. Xiang Gao (高翔)
Ph. D of Tsinghua University
Postdoc of Technical University of Munich
```