gaoxiang12/faster-lio #41: 有关重力对齐的问题

hr2894235132, 2022-11-02:
高博您好！首先非常感谢您这么优秀的工作。想问一下faster-lio中是否做了像vins-mono中重力对齐的处理呢？我在看了一遍代码后并没有找到。

gaoxiang12, 2022-11-02:
你好，fasterlio里重力方向是在线估计的，一开始旋转矩阵为I，重力方向会在球面上进行调整，跑久了之后可能和一开始不在同一个方向上。
vins里面具体怎么做的我不太清楚，大部分应该是初始化时估计旋转矩阵，然后重力固定为9.8这样。

hr2894235132, 2022-11-02:
> 你好，fasterlio里重力方向是在线估计的，一开始旋转矩阵为I，重力方向会在球面上进行调整，跑久了之后可能和一开始不在同一个方向上。 vins里面具体怎么做的我不太清楚，大部分应该是初始化时估计旋转矩阵，然后重力固定为9.8这样。

想请问一下重力方向的在线估计具体是哪一块呢。只找到了下面这行代码：
`init_state.grav = S2(-mean_acc_ / mean_acc_.norm() * common::G_m_s2);`
vins里好像是是补偿了yaw角，进行重力方向的对齐。

gaoxiang12, 2022-11-02:
在线估计就是说运动和观测里都会有g相关的项。
fastlio系列是把重力看作球面矢量，然后对这个球进行旋转来更新重力的方向。如果你打印整个slam过程中的重力估计值，它是会变的。具体的代码就和eskf观测模型放在一起了。

hr2894235132, 2022-11-03:
> 

好的，非常感谢您的帮助和解答，我自己再学习一下，再次感谢！祝您工作顺利，生活愉快！

closed 2022-11-07 by gaoxiang12

hr2894235132, 2022-11-08:
> 在线估计就是说运动和观测里都会有g相关的项。 fastlio系列是把重力看作球面矢量，然后对这个球进行旋转来更新重力的方向。如果你打印整个slam过程中的重力估计值，它是会变的。具体的代码就和eskf观测模型放在一起了。

高博您好！很抱歉再次打扰您，我想问一下，如果我想让雷达坐标系和重力方向对齐（也就是让z轴一直竖直向下），在初始化的时候估计旋转矩阵，进行补偿，那这个初始的雷达坐标系是和imu系一致的吗？雷达坐标系是什么呢？