gaoxiang12/faster-lio #43: 关于论文中的LIO-SAM 的 nclt_2 指标问题

xingyuuchen, 2022-11-28:
Hello, 首先感谢您们的工作！

我有一个关于您们如何得到 LIO-SAM 在 `nclt_2` 上的指标的小问题（指标如下）：
[image]

我根据您们的`NCLT`配置文件 [velodyne.yaml](https://github.com/gaoxiang12/faster-lio/blob/main/config/velodyne.yaml) 更改了LIO-SAM的配置 [params.yaml](https://github.com/TixiaoShan/LIO-SAM/blob/master/config/params.yaml)，具体而言我做了如下更改:

1. `scan_line`: 对应 LIO-SAM 配置文件中的 `N_SCAN` 设置到`32`;
2. `extrinsic_T`: 对应 LIO-SAM 配置文件中的 `extrinsicTrans` 设置到 `[ 0, 0, 0.28 ]`;
3.  IMU的方差与对应的随机游走方差也根据您们的 NCLT 配置文件做了相应更改;
4. `time_scale`: 由于 LIO-SAM 在 deskew 的时候使用`秒`做单位, 而我注意到 `nclt_2` (i.e. `20130110.bag`) 提供的是 `us`，因此每个点相对其当前scan第一个点的时间除以了`1e6`（我确保了每个scan最后一个点的时间是 0.1s，这和LIDAR的10Hz一致）;

_如果不做上述第三点更改_，结果如下（可以看到走出最开始的那个平台（像是一个停车场）后，整个场景就倾斜了）:
[image]
使用 `evo` 测出的误差指标（对齐轨迹后）也比论文中的高出一个数量级（不对齐会更高）。

_如果做了第三点更改_，大约刚刚走出最开始的平台，轨迹就完全飞了。

由于LIO-SAM没有在 `NCLT` 数据集上实验，所以请允许我在此请教。

我想知道我是否漏掉了某个配置或改动，恳请指导～

谢谢！

xingyuuchen, 2022-11-29:
同时我观察到FAST-LIO2中的这个指标是7.9:

[image]
[image]

closed 2023-03-08 by xingyuuchen

liukeq, 2023-11-19:
> Hello, 首先感谢您们的工作！
> 
> 我有一个关于您们如何得到 LIO-SAM 在 `nclt_2` 上的指标的小问题（指标如下）： [image]
> 
> 我根据您们的`NCLT`配置文件 [velodyne.yaml](https://github.com/gaoxiang12/faster-lio/blob/main/config/velodyne.yaml) 更改了LIO-SAM的配置 [params.yaml](https://github.com/TixiaoShan/LIO-SAM/blob/master/config/params.yaml)，具体而言我做了如下更改:
> 
> 1. `scan_line`: 对应 LIO-SAM 配置文件中的 `N_SCAN` 设置到`32`;
> 2. `extrinsic_T`: 对应 LIO-SAM 配置文件中的 `extrinsicTrans` 设置到 `[ 0, 0, 0.28 ]`;
> 3. IMU的噪声、随机游走方差也根据您们的 NCLT 配置文件做了相应更改;
> 4. `time_scale`: 由于 LIO-SAM 在 deskew 的时候使用`秒`做单位, 而我注意到 `nclt_2` (i.e. `20130110.bag`) 提供的是 `us`，因此每个点相对其当前scan第一个点的时间除以了`1e6`（我确保了每个scan最后一个点的时间是 0.1s，这和LIDAR的10Hz一致）;
> 
> _如果不做上述第三点更改_，结果如下（可以看到走出最开始的那个平台（像是一个停车场）后，整个场景就倾斜了）: [image] 使用 `evo` 测出的误差指标（对齐轨迹后）也比论文中的高出一个数量级（不对齐会更高）。
> 
> _如果做了第三点更改_，大约刚刚走出最开始的平台，轨迹就完全飞了。
> 
> 由于LIO-SAM没有在 `NCLT` 数据集上实验，所以请允许我在此请教。
> 
> 我想知道我是否漏掉了某个配置或改动，恳请指导～
> 
> 谢谢！

请问最后你们最后怎么解决的？我也遇到了这些问题

xingyuuchen, 2023-11-19:
> 请问最后你们最后怎么解决的？我也遇到了这些问题

没有解决，他数据就是对不上。。。

YuexinMu, 2024-10-26:
请问最后你们最后解决了吗？我也遇到了这个问题