gaoxiang12/faster-lio #45: Computation and memory usage surge after running for a long time (10-20 mins)

XuRobotics, 2022-12-31:
Thanks for releasing this powerful LIDAR odometry stack! 

We have been trying to incorporate this into our autonomy stack (see here: https://github.com/KumarRobotics/kr_autonomous_flight), and performing lots of experiments onboard our UAVs. We have been very impressed by the accuracy and robustness of this odometry system. However, we did observe some abnormal computation and memory surges when we performed long-duration (10-20 mins) operations. 

The LDIAR odometry takes up to 16 GB of memory, and up to 4-6 threads of our 12-thread intel i7-10710U processor. As a result, the frequency of the odometry sometimes reduces to below 10Hz (sometimes the odometry stops outputting any estimates for more than 1-2 seconds), which leads to the unstable behaviors of the UAV. This problem is especially severe when the LIDAR stays at the same location for a long time.

**I would like to know if this is expected behavior. If yes, is there any change to the code that I can make to alleviate the problem? If not, we can help look into the potential causes, could you please provide any suggestions on where to debug?**

More details that might be helpful:

We are using an Intel NUC as our onboard computer (model intel-nuc10i7fnk with [Intel® Core™ i7-10710U Processor (12M Cache, up to 4.70 GHz)] https://www.intel.com/content/www/us/en/products/sku/196448/intel-core-i7-10710u-processor-12m-cache-up-to-4-70-ghz/specifications.html)).

We are using ouster OS1-64 (gen 2) as our sensor and the default ouster64.yaml file in this stack as the configuration.

Yuea922, 2024-07-22:
I have also noticed this issue, may I ask if there is any progress

Freeskylover, 2025-03-19:
[image]

because this code is not same with origin fastlio2, when faster-lio is updating map, it not sure that one voxel one point,  so result is that  one voxel accumulate many points, finally the ram run out quickly.

Freeskylover, 2025-03-19:
[image]
origin fastlio2 downsample the "points_to_add points" voxel, sure that one voxel one point.