gaoxiang12/faster-lio #57: Do you have any plans to make rosbag public without requiring specific access right?

mu-mu-mu, 2023-07-24:
Thank you for your great work.
I am currently attempting to replicate your work on my local machine for better understanding.
The rosbag links listed on [prepare-the-dataset](https://github.com/gaoxiang12/faster-lio#prepare-the-datasets)  seem to require specific access rights.
Would it be possible for you to make these links accessible?
I would like to download them via command line tools such as gdown.
Thank you again!

gaoxiang12, 2023-07-25:
Hi @mu-mu-mu 
Actually, these links are not owned by me so I cannot set the access for you. But you can try this link and it also has the NCLT, UTBM bags: https://1drv.ms/u/s!AgNFVSzSYXMahcEZejoUwCaHRcactQ?e=YsOYy2  

Feel free to let me know if I can provide any other help.

mu-mu-mu, 2023-07-26:
I could download the rosbags via the link and the faster-lio worked :)
Thank you for your help!

closed 2023-07-26 by mu-mu-mu