gaoxiang12/faster-lio #60: 对fastlio2和fasterlio的近邻搜索发现ivox搜索时间长

seanXMU, 2023-08-23:
高博您好，
我在测试两者的搜索时间时，测试了avia的outdoor_Mainbuilding_100Hz_2020-12-24-16-46-29.bag ，发现ikdtree的搜索时间小于ivox，和论文的knn实验比较有矛盾。fasterlio和fastlio2的相关参数一样，代码修改的仅有在Nearest_Search函数前后获取时间，请问这里是什么问题呢？

[image]
上图是两者的时间，下图蓝色代表ivox和ikdtree的搜索时间差。

gaoxiang12, 2023-08-23:
这也差太多了，查下是否有编译上的问题。

seanXMU, 2023-08-23:
图里的单位是微秒，ivox的搜索最多慢个500us，但是faster-lio的update的时间又明显快了很多。所以现在测的情况就是h_share_model函数里面的搜索部分比较慢，但是在update整体的时间又比较快。很懵，请问faster-lio里面在update这部分有除了h_share_model以外的修改吗

gaoxiang12, 2023-08-23:
我并不知道你那边是怎么测的。。我之前是在fasterlio框架下弄的，做了个单元测试

seanXMU, 2023-08-28:
我在`ObsModel`的`ivox_->GetClosestPoint(point_world, points_near, options::NUM_MATCH_POINTS);`的前后分别获取t1,t2然后做差保存数据，请问这样会有什么问题吗？