gaoxiang12/faster-lio #64: 关于数据包bag无法访问的问题

MintonLee, 2023-10-08:
您好,我的账号无法访问bag包的数据,我的google email是:   liming19960210@gmail.com

youngsterkim, 2023-10-09:
您好,我的账号无法访问bag包的数据，我的email是：wanghuohuocy@gmail.com。谢谢!!!!

gaoxiang12, 2023-10-09:
Hello，数据包并不在我的google帐号下，所以您可能需要向fastLio原作者申请。
或者，也可以通过SAD书里的链接，下载NCLT数据包：

百度云链接: https://pan.baidu.com/s/1ELOcF1UTKdfiKBAaXnE8sQ?pwd=feky 提取码: feky
OneDrive链接：https://1drv.ms/u/s!AgNFVSzSYXMahcEZejoUwCaHRcactQ?e=YsOYy2

MintonLee, 2023-10-09:
> Hello，数据包并不在我的google帐号下，所以您可能需要向fastLio原作者申请。 或者，也可以通过SAD书里的链接，下载NCLT数据包：
> 
> 百度云链接: https://pan.baidu.com/s/1ELOcF1UTKdfiKBAaXnE8sQ?pwd=feky 提取码: feky OneDrive链接：https://1drv.ms/u/s!AgNFVSzSYXMahcEZejoUwCaHRcactQ?e=YsOYy2

谢谢高博士

youngsterkim, 2023-10-09:
> Hello，数据包并不在我的google帐号下，所以您可能需要向fastLio原作者申请。 或者，也可以通过SAD书里的链接，下载NCLT数据包：
> 
> 百度云链接: https://pan.baidu.com/s/1ELOcF1UTKdfiKBAaXnE8sQ?pwd=feky 提取码: feky OneDrive链接：https://1drv.ms/u/s!AgNFVSzSYXMahcEZejoUwCaHRcactQ?e=YsOYy
太感谢您了