gaoxiang12/faster-lio #67: IVox第一帧点云坐标系问题

YWL0720, 2023-12-05:
高博您好，感谢您的工作，在学习您的代码时，有这样的一个问题。按照我的理解，faster-lio估计的是IMU坐标系在世界坐标系下的状态，而世界坐标系应该与第一帧IMU坐标系重合。但在函数 **LaserMapping::Run()** 中，处理第一帧点云时，却直接将去畸变后，在雷达坐标系下的点云 **scan_undistort_** 加入到了ivox中 
```cpp
/// the first scan
    if (flg_first_scan_) {
        ivox_->AddPoints(scan_undistort_->points);
        first_lidar_time_ = measures_.lidar_bag_time_;
        flg_first_scan_ = false;
        return;
    }
```
这里是否应该也和后续的地图更新函数 **LaserMapping::MapIncremental()** 中的做法一致，将点云转换到世界坐标系下再加入到ivox中呢？
```cpp
PointBodyToWorld(&(scan_down_body_->points[i]), &(scan_down_world_->points[i]));
```
期待着您的答复。

YWL0720, 2023-12-08:
高博您好，我做了个简单的实验验证了这个问题，如果有分析不对的地方还希望您能给出指正。
首先我注释掉了 **laser_mapping.cc** 中的第323行
```cpp
// Timer::Evaluate([&, this]() { MapIncremental(); }, "    Incremental Mapping");
```
即不再进行地图更新，地图中仅保存第一帧点云。正常情况下仅依靠第一帧点云，在雷达移动距离较近时，也会约束住。应该表现为能维持一段时间的稳定位姿估计，当新的点云与第一帧点云共同区域较小时位姿估计失败。
接下来我分别进行了三次测试
### 1. UrbanLoco Hong Kong Dataset Dataset 5: HK-Data20190117 
```bash
roslaunch faster_lio mapping_velodyne_ulhk.launch
```
现象是直接沿横向飞走。
[image]

### 2. utbm utbm_robocar_dataset_20180713_noimage.bag
```bash
roslaunch faster_lio mapping_velodyne_utbm.launch
```
现象是平移震荡后维持了一段时间的稳定的估计。
[image]

### 3. hku_main_building.bag
```bash
roslaunch faster_lio mapping_avia.launch
```
现象是持续了一段时间的稳定估计，雷达与第一帧点云无共同视野时飘走。

### 4. 分析
- ulhk数据集中雷达坐标系与IMU坐标系之间存在较大的旋转角度偏差，由于该角度偏差，导致随后世界坐标系下的点云无法与第一帧雷达坐标系下的点云配准成功，进而导致飞走。
```
extrinsic_T: [ 0, 0, -0.28]
extrinsic_R: [ 0, -1,  0, 1,  0,  0, 0,  0,  1]
```
- utbm数据集中雷达坐标系与IMU坐标系之间外参平移分量较大，而旋转保持相同，因而能在几帧之内将位姿纠正，体现为平移震荡。
```
extrinsic_T: [ -0.5, 1.4, 1.5]
extrinsic_R: [ 1, 0, 0,  0, 1, 0,   0, 0, 1]
```
- hku_main_building数据集使用avia雷达，其内置IMU，雷达坐标系和IMU坐标系在旋转和平移上都没有太大差异，因而测试起来效果最合乎预期。
```
extrinsic_T: [ 0.04165, 0.02326, -0.0284 ]
extrinsic_R: [ 1, 0, 0, 0, 1, 0,  0, 0, 1 ]
```
### 5. 补充实验
我将第一帧的点云转换到世界坐标系下后，再加入到IVox中，重复上述实验，均为正常表现，即维持了一段时间的稳定位姿估计，当新的点云与第一帧点云共同区域较小时位姿估计失败。

### 6. 总结
根据上面的实验，我还是认为第一帧点云应当转换到世界坐标系下后再加入到IVox中。虽然实际上由于每次估计后会对地图进行更新，新的点云很快会覆盖掉第一帧点云，导致最终的差异并不明显。

期待着您的回复。

gaoxiang12, 2025-03-17:
感谢，请给我提一个PR好吗？

mainDevgithub, 2025-03-18:
又用一台性能更好的台式机进行了对比，所用数据是 utbm_20190418_noimage.bag

发现在台式机上faster-lio 能稳定运行；即使改参数，faster-lio 依旧会在原先的笔记本电脑上跑飞（初始振荡，后来飞掉）。两台电脑都是 ubuntu18.04，ros 版本相同，会不会是性能的问题？但用同样的数据，笔记本电脑能稳定运行 fast-lio2

1. 台式机 & faster-lio

初始也很稳（后面 fast-lio2 这里显示的两个坐标系好像是完全重合的状态）
[image]

[image]

[image]

2. 笔记本 & faster-lio

初始振荡
[image]

很快飞掉
[image]

3. 笔记本 & fast-lio2 一直很稳

[image]

[image]

[image]