gaoxiang12/faster-lio #74: 同一个rosbag每一次运行的结果都不一样

HP-Yin, 2024-03-29:
高博你好，请问一下，我用faster-lio跑rosbag数据为什么每次运行的结果都有一些细微的区别呢，比如位置都有一两厘米的差异（如图所示打印出来了），而fast-lio就没有这样的情况，是因为ivox的一些机制造成的吗，还是说有一些参数需要修改呢。数据是用Livox Avia录制的。
[image]

gaoxiang12, 2024-03-30:
实时跑的吗？还是录下来之后用离线来跑的？

HP-Yin, 2024-03-30:
rosbag录下来之后跑的，用的是LivoxAVIA雷达，一直都跑的同一个rosbag。排查了一下，其中一个原因是预处理的AviaHandler函数中使用了std::for_each(std::execution::par_unseq, index.begin(), index.end(), [&](const uint &i)，采用并行方法的遍历可能会有随机性导致返回的cloud_out_的size每次都不同。这边改成用for循环之后还是有同样的现象，正在排查。

GavyndotLee, 2024-04-28:
@HP-Yin 你好，我也遇到了相同的现象，也是用rosbag录制的数据跑的。请问定位到原因了吗？

HP-Yin, 2024-04-28:
感觉影响不大，你可以把代码里的std::execution::unseq和std::execution::par_unseq都改用for循环试试

HP-Yin, 2024-04-28:
@GavyndotLee 感觉影响不大，你可以把代码里的std::execution::unseq和std::execution::par_unseq都改用for循环试试

GavyndotLee, 2024-04-28:
@HP-Yin 谢谢你的回复，改成for循环之后每次运行的结果仍然不一样。我使用的是robosense16录制的室内数据，x、y变化超过1米了。

GavyndotLee, 2024-04-28:
@gaoxiang12 高博你好，我使用robosense16录制了一个室内的bag文件，用在线的方式运行，然后播放bag包。faster-lio每一次运行的结果x、y变化超过1米，按照HP-Yin的建议将std::execution::unseq、std::execution::par_unseq改为普通for循环后仍然存在这个现象。我测试了fast-lio2每次运行结果是一致的。高博能否提供一些解决方案或者思路，感谢！

HP-Yin, 2024-07-09:
@GavyndotLee 主要改预处理的和 obs_model 里面的遍历，我试过了，问题可以解决

huangwentao123, 2024-08-07:
请问具体在源码里是哪些文件的那几行呢，我修改后没有效果，反而结果变差了很多 直接飘了

Freeskylover, 2025-02-05:
> [@gaoxiang12](https://github.com/gaoxiang12) 高博你好，我使用robosense16录制了一个室内的bag文件，用在线的方式运行，然后播放bag包。faster-lio每一次运行的结果x、y变化超过1米，按照HP-Yin的建议将std::execution::unseq、std::execution::par_unseq改为普通for循环后仍然存在这个现象。我测试了fast-lio2每次运行结果是一致的。高博能否提供一些解决方案或者思路，感谢！

在跑fastlio2也有这个问题，同一个数据包结果不一样。