gaoxiang12/faster-lio #82: [laserMapping-2] process has died

leo112578, 2024-07-13:
[laserMapping-2] process has died [pid 1258, exit code -6, cmd /home/yzh/catkin_ws/devel/lib/faster_lio/run_mapping_online __name:=laserMapping __log:=/root/.ros/log/92cef308-4111-11ef-80f7-e88088734254/laserMapping-2.log].
log file: /root/.ros/log/92cef308-4111-11ef-80f7-e88088734254/laserMapping-2*.log

leo112578, 2024-07-13:
请问有遇到过相同问题的吗？是怎么解决的啊？

dayeone, 2024-08-28:
am also experiencing the same issue. 
Were you able to resolve it? 
If so, I would really appreciate it if you could share the solution with me.

dayeone, 2024-08-28:
I am also experiencing the same issue.
Were you able to resolve it? 
I would really appreciate it if you could share the solution with me.

vzhr, 2025-07-17:
在pointcloud_preprocess.cc中，VelodyneHandler函数没有对nan处理，加入如下：
`void PointCloudPreprocess::VelodyneHandler(const sensor_msgs::PointCloud2::ConstPtr &msg) {
    cloud_out_.clear();
    cloud_full_.clear();

    pcl::PointCloud<velodyne_ros::Point> pl_orig0, pl_orig;
//    LOG(INFO) << "from msg";
    pcl::fromROSMsg(*msg, pl_orig0);
    std::vector<int> index;
    pcl::removeNaNFromPointCloud<velodyne_ros::Point>(pl_orig0, pl_orig, index);
//    LOG(INFO) << "from msg end";
    int plsize = pl_orig.points.size();
    if (plsize == 0)
    {
        return;
    }
    cloud_out_.reserve(plsize);`