gaoxiang12/faster-lio #85: Using with Livox Mid-360

gmdeoliveira, 2024-09-10:
Do you have any tips to run this algorithm using with Livox Mid-360?

LIN-DN, 2024-10-15:
I also have this problem, have you solved it now? Can you share the config and YAML files related to mid360

allenthreee, 2025-01-09:
> I also have this problem, have you solved it now? Can you share the config and YAML files related to mid360

change lidar type of AVIA.yaml to 2(velodyne), MID360 output is /sensor_msg/pointcloud2, which is same as velodne and ouster

gmdeoliveira, 2025-01-15:
Hello everyone, I could run the Faster-LIO just doing some adjustments in the YAML file, follows the link: https://github.com/gmdeoliveira/faster-lio/blob/main/config/mid360_test.yaml

ereshkigal224, 2025-01-18:
> Hello everyone, I could run the Faster-LIO just doing some adjustments in the YAML file, follows the link: https://github.com/gmdeoliveira/faster-lio/blob/main/config/mid360_test.yaml

thank u for ur contribution

ZTL0806, 2025-04-19:
https://blog.csdn.net/qq_64931749/article/details/147287340?fromshare=blogdetail&sharetype=blogdetail&sharerId=147287340&sharerefer=PC&sharesource=qq_64931749&sharefrom=from_link我写了一篇使用mid360来进行仿真的帖子，可以参考一下